Smanar/CYD-Domoticz-Remote #1: Questions regarding Button and Rain Sensor

franki29, 2024-02-22:
Hi there,

what a great software, thanks for providing it.

I have two questions:

1.) Possible to have less Buttons (Like 6 ) but bigger ones?

2.) Is it possible to add Rain Sensor? Here a Json extract:
Protected	false
Rain	"1.2"
RainRate	"2223.6"
ShowNotifications	true
SignalLevel	"-"
SubType	"TFA"
Timers	"false"
Type	"Rain"
TypeImg	"rain"
Unit	1
Used	1
XOffset	

Br. Frank

Smanar, 2024-02-22:
The problem is all users want a different homepage ^^.
The code is dynamic, have you tried to set 
```
	-DTOTAL_ICONX=3 # How many icon widht
	-DTOTAL_ICONY=2 # How many icon Heigh
```
for exemple ?

I m adding the rain sensor, but what is the "data" value on the json, I m using this one for all others sensors, strange this one use "RainRate "

And take care the actual version have memory issues (not stable), I m on it ATM

franki29, 2024-02-23:
Hi, thanks for your answer. Icon is understood. And it is working at my side.
After update to newest version Rain is now displayed but unfortunately the  Rain rate, which is the total count. I would like to have the current on which will be Rain. (In my example 1.2) Where can i change it in the code?

Smanar, 2024-02-23:
>Where can i change it in the code?

Don't worry I can edit it, it will be fast. My problem is to find the value. I have created a "fake" sensor and all values are in "data" on my side (even they are too in "Rain" and "RainRate".
It's not same for you ?
If I m right as value you need to have x;y with x is the "Rainrate" and y the "rain" ?

franki29, 2024-02-23:
But I am pretty sure that this is not the right way to do

franki29, 2024-02-23:
For Rain, Temperature , Humidity , Pressure and Wind I have a Bresser Weather Station. The data are pushed by the https://github.com/matthias-bs/BresserWeatherSensorReceiver

Smanar, 2024-02-23:
Thx, I will take a look tommorow
But what is the "data" value ? It's not possible to use it ? (Because at least 99.99% of domoticz widget just use this field)

franki29, 2024-02-24:
Hi, no both are present:

Rain "1.2"
RainRate "2223.6"

Smanar, 2024-02-24:
But can you show me your "data" field, this one is missing on your extract
```
Protected false
Rain "1.2"
RainRate "2223.6"
ShowNotifications true
SignalLevel "-"
SubType "TFA"
Timers "false"
Type "Rain"
TypeImg "rain"
Unit 1
Used 1
XOffset
```

I have tried to make a fake rain sensor, but I have not the same than you, mine have all value in "data"

franki29, 2024-02-25:
[image]

Hi, it looks like this and here is the json:
{
	"ActTime" : 1708880237,
	"AstrTwilightEnd" : "19:56",
	"AstrTwilightStart" : "05:37",
	"CivTwilightEnd" : "18:39",
	"CivTwilightStart" : "06:54",
	"DayLength" : "10:38",
	"NautTwilightEnd" : "19:18",
	"NautTwilightStart" : "06:15",
	"ServerTime" : "2024-02-25 17:57:17",
	"SunAtSouth" : "12:47",
	"Sunrise" : "07:28",
	"Sunset" : "18:05",
	"app_version" : "2024.4",
	"result" : 
	[
		{
			"AddjMulti" : 1.0,
			"AddjMulti2" : 1.0,
			"AddjValue" : 0.0,
			"AddjValue2" : 0.0,
			"BatteryLevel" : 255,
			"CustomImage" : 0,
			"Data" : "223920.0;2239.2",
			"Description" : "",
			"Favorite" : 1,
			"HardwareDisabled" : false,
			"HardwareID" : 3,
			"HardwareName" : "Wohnzimmer Lampe",
			"HardwareType" : "Dummy (Does nothing, use for virtual switches only)",
			"HardwareTypeVal" : 15,
			"HaveTimeout" : false,
			"ID" : "143BE",
			"LastUpdate" : "2024-02-25 17:57:11",
			"Name" : "Regen",
			"Notifications" : "false",
			"PlanID" : "0",
			"PlanIDs" : 
			[
				0
			],
			"Protected" : false,
			"Rain" : "0.8",
			"RainRate" : "2239.2",
			"ShowNotifications" : true,
			"SignalLevel" : "-",
			"SubType" : "TFA",
			"Timers" : "false",
			"Type" : "Rain",
			"TypeImg" : "rain",
			"Unit" : 1,
			"Used" : 1,
			"XOffset" : "0",
			"YOffset" : "0",
			"idx" : "878"
		}
	],
	"status" : "OK",
	"title" : "Devices"
}

Smanar, 2024-02-25:
`"Data" : "223920.0;2239.2",
"Rain" : "0.8",
"RainRate" : "2239.2",`

>The device only send the total amount of rain as far I know, the rest must be calculated

Yes I m seing.

```
/json.htm?type=command&param=udevice&idx=IDX&nvalue=0&svalue=RAINRATE;RAINCOUNTER

    IDX = id of your device (This number can be found in the devices tab in the column "IDX")
    RAINRATE = amount of rain in last hour in [mm x 100]
    RAINCOUNTER = continues counter of fallen Rain in [mm]
```

So data contain only Rainrate and total counter, and you need the "today rain" ? So no choice, need to make like you have done.
I will try to update the code today, else you will have it tommorow.

Smanar, 2024-02-27:
Ok, so have corrected a big issue, with the help from the arduinojson lib dev.
Have added a "hack" for this sensor, the data will now contain 3 fields, the total rain, the rain rate and the today rain.

If you want to  test it, the code is atm on the "test" branch, so much critic code to be pushed on master immedialty.

Smanar, 2024-03-01:
Have corrected somethinh yesterday, grpah was not working, but still on "test" branch.

franki29, 2024-03-03:
Hi I tested the newest version where you insert the special device for Rain.
Output is now :
 224280.0
2242.8
0.0

But I need only the 0.0. (Sunny Day :-) )

So I add:

 if (RJson2.containsKey("Rain"))
{
    data = Cleandata(JSondata, RJson2["Rain"]);

    // Assuming data is a char* representing the cleaned data

    // Find the position of the second newline character
    char *startIndex = strchr(data, '\n');
    if (startIndex != NULL)
    {
        startIndex = strchr(startIndex + 1, '\n');
        if (startIndex != NULL)
        {
            // Extract the substring after the second newline
            startIndex++; // Move startIndex to the character after the second newline
            Serial.println(startIndex);
            data=startIndex;
        }
    }
}

and 
if (i.containsKey("Rain"))
{
    data = Cleandata(JSondata, i["Rain"]);
    // Assuming data is a char* representing the cleaned data

    // Find the position of the second newline character
    char *startIndex = strchr(data, '\n');
    if (startIndex != NULL)
    {
        startIndex = strchr(startIndex + 1, '\n');
        if (startIndex != NULL)
        {
            // Extract the substring after the second newline
            startIndex++; // Move startIndex to the character after the second newline
            Serial.println(startIndex);
            data = startIndex;
        }
    }
}

To get only the third entry, but again with help from the AI

Smanar, 2024-03-03:
But the problem is another user might want another field, it's not good if you have all of them ?

franki29, 2024-03-04:
In case of the bresser sensor I think it make no big sense as you do not know at what date the sensor was starting for collection of rain data. The calculation of rain per hour is made by Domoticz.

Of course with other rain sensor it might be good to have it.

Smanar, 2024-03-04:
> In case of the bresser sensor I think it make no big sense as you do not know at what date the sensor was starting for collection of rain data. The calculation of rain per hour is made by Domoticz.
> 
> Of course with other rain sensor it might be good to have it.

Ha yes, not wrong, a counter is stupid for that except if you reset it manualy at some period .....
Honnestly IDK what to do, lol.
As you are the first one that is trying this sensor, there is something easier to code, replace

```
    if (RJson2.containsKey("Rain"))
    {
        data = Cleandata(JSondata, RJson2["Rain"]);
    }
```

by

```
    if (RJson2.containsKey("Rain"))
    {
        data = Cleandata(RJson2["Rain"]);
    }
```

You will have only 1 value, but the good one.
Need to do same for

```
        if (i.containsKey("Rain"))
        {
           data = Cleandata(JSondata, i["Rain"]);
        }
```

The last code version is on branch "test", I have just updated it, so this change will be in next version.

franki29, 2024-03-29:
Hi, in the main release the graph for rain does not work anymore. Here are the data: (Graph chows only 0 for all data))

<div class="toolbar"><div class="devtools-separator"></div><div class="devtools-searchbox"><input class="searchBox devtools-filterinput" placeholder="JSON durchsuchen" value=""></div></div><div class="panelContent">
  |  
-- | --
result |  
0 |  
d | "2024-03-28 16:00"
mm | "0.0"
1 |  
d | "2024-03-28 17:00"
mm | "0.0"
2 |  
d | "2024-03-28 18:00"
mm | "0.0"
3 |  
d | "2024-03-28 19:00"
mm | "0.0"
4 |  
d | "2024-03-28 20:00"
mm | "0.0"
5 |  
d | "2024-03-28 21:00"
mm | "0.0"
6 |  
d | "2024-03-28 22:00"
mm | "0.0"
7 |  
d | "2024-03-28 23:00"
mm | "0.0"
8 |  
d | "2024-03-29 00:00"
mm | "0.0"
9 |  
d | "2024-03-29 01:00"
mm | "0.0"
10 |  
d | "2024-03-29 02:00"
mm | "0.0"
11 |  
d | "2024-03-29 03:00"
mm | "0.0"
12 |  
d | "2024-03-29 04:00"
mm | "0.0"
13 |  
d | "2024-03-29 05:00"
mm | "0.8"
14 |  
d | "2024-03-29 06:00"
mm | "0.4"
15 |  
d | "2024-03-29 07:00"
mm | "0.0"
16 |  
d | "2024-03-29 08:00"
mm | "0.0"
17 |  
d | "2024-03-29 09:00"
mm | "0.0"
status | "OK"
title | "Graph rain day"

</div>

Smanar, 2024-03-29:
I think it's because values are too low, all < 1 and decimal.
Have updated the code on the "test" branch.

franki29, 2024-04-01:
Great works, thank you very much

closed 2024-04-01 by franki29