Smanar/CYD-Domoticz-Remote #4: Support for ESP32-4848S040

franki29, 2024-03-08:
Hi, I just bought a ESP32-4848S040

https://github.com/Mboukhal/4.0inch_ESP32-4848S040

https://github.com/arendst/Tasmota/discussions/20527

Do you think you can make it work for your Software?

Br. Frank

Smanar, 2024-03-09:
Why not.
So it seem there is more than 1 version of this device, with less or more relay.

CPU : ESP32-S3-WROOM-1-MCN16R8
Flash : 16M
Psram : 8 M
Resolution 480*480
Display interface : Direct 16 bits + 3 wire SPI
Controller : ST7701
Touchpad : GT911

On github :
- TFT_eSPI : No project found
- LovyanGFX : https://github.com/Mboukhal/4.0inch_ESP32-4848S040/blob/master/1-Demo/Demo_Arduino/1_1_86switch_onoff/86switch_onoff/main.cpp1
- ArduinoGFX, I think it's the lib the more used for this device.

On your link we can see the dev have tried with LovyanGFX and have give up to finaly try with ArduinoGFX.
So without the device, better to make with working code.

I m rewrited the code to use the ArduinoGFX library. but this lib is not the faster one. If you can found a project with your device but using an other library will be usefull.

For the moment I have a c++ problem about the synthax to keep the code multi support. I m asking some help to his github

franki29, 2024-03-09:
Hi found a other project that seems to work:

https://github.com/HASwitchPlate/openHASP

Smanar, 2024-03-09:
Yep, have found this one too, and they are using too ArduinoGFX.
I m waiting an anwser from their dev, but I think the support is possible at least with this lib.

Smanar, 2024-03-10:
I have started something if you can make a try.
On the branch "test".
On the platoform.ini file, edit the code for your device
```
default_envs = esp32-2432S028R
;default_envs = esp32-2432S024C
;default_envs = esp32-E32S3RGB43
;default_envs = esp32-4848S040
```

The touchpad is not enabled yet.
But with some luck you will see the calibration screen.

Smanar, 2024-03-11:
Ha ? you have a crash after the screen setup.
So after the calibration screen, without the touchpad all was done in less than 1s, so not possible to see if it work.

I have pushed a new version.
- The arduno GFX code work, tested on my device
- Have started a touchpad support
- Have added more debug line to see where the application crash https://github.com/Smanar/CYD-Domoticz-Remote/commit/2912b3465f6270c3b9dc11f533a4358e70d031a8

Same than before the code is for the esp32-2432S028R, so need to comment/uncomment
```
;default_envs = esp32-2432S028R
;default_envs = esp32-2432S024C
;default_envs = esp32-E32S3RGB43
default_envs = esp32-4848S040
```

franki29, 2024-03-12:
Hi,
of course I change the defaul_envs.
With the new code it is rebooting continious and  I get:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1

Smanar, 2024-03-14:
Hello, not enought time ? or need more explanation  ?

franki29, 2024-03-14:
Sorry not enough time, will test tomorrow

franki29, 2024-03-15:
Hi I tried with disabeling the ;-DTOUCH_911, but than I get compiling errors:
src/core/screen_driver.cpp: In function 'void touchscreen_calibrate(bool)':
src/core/screen_driver.cpp:244:5: error: 'TS_Point' was not declared in this scope
     TS_Point p;
     ^~~~~~~~
src/core/screen_driver.cpp:244:5: note: suggested alternative: 'Print'
     TS_Point p;
     ^~~~~~~~
     Print
src/core/screen_driver.cpp:247:12: error: 'touchscreen' was not declared in this scope
     while (touchscreen.touched())
            ^~~~~~~~~~~
src/core/screen_driver.cpp:247:12: note: suggested alternative: 'touchRead'
     while (touchscreen.touched())
            ^~~~~~~~~~~
            touchRead
src/core/screen_driver.cpp:251:13: error: 'touchscreen' was not declared in this scope
     while (!touchscreen.touched())
             ^~~~~~~~~~~
src/core/screen_driver.cpp:251:13: note: suggested alternative: 'touchRead'
     while (!touchscreen.touched())
             ^~~~~~~~~~~
             touchRead
src/core/screen_driver.cpp:254:5: error: 'p' was not declared in this scope
     p = touchscreen.getPoint();
     ^
src/core/screen_driver.cpp:254:9: error: 'touchscreen' was not declared in this scope
     p = touchscreen.getPoint();
         ^~~~~~~~~~~
src/core/screen_driver.cpp:254:9: note: suggested alternative: 'touchRead'
     p = touchscreen.getPoint();
         ^~~~~~~~~~~
         touchRead
src/core/screen_driver.cpp: In function 'void screen_lv_touchRead(lv_indev_drv_t*, lv_indev_data_t*)':
src/core/screen_driver.cpp:405:5: error: 'TS_Point' was not declared in this scope
     TS_Point p;
     ^~~~~~~~
src/core/screen_driver.cpp:405:5: note: suggested alternative: 'Print'
     TS_Point p;
     ^~~~~~~~
     Print
src/core/screen_driver.cpp:407:9: error: 'touchscreen' was not declared in this scope
     if (touchscreen.tirqTouched() && touchscreen.touched())
         ^~~~~~~~~~~
src/core/screen_driver.cpp:407:9: note: suggested alternative: 'touchRead'
     if (touchscreen.tirqTouched() && touchscreen.touched())
         ^~~~~~~~~~~
         touchRead
src/core/screen_driver.cpp:420:9: error: 'p' was not declared in this scope
         p = touchscreen.getPoint();
         ^
*** [.pio\build\esp32-4848S040\src\core\screen_driver.cpp.o] Error 1

Smanar, 2024-03-15:
Ha yes you are right, have pushed a new version with touch disabled for your device

```
	;-DTOUCH_911
	-DTOUCH_DISABLED
```

You can re-enalbe it later with switching the comment.

franki29, 2024-03-15:
I tried it, but now there is not even a serial output

Smanar, 2024-03-15:
Even with a restart ?
Or you have direclty the crash message ?

Because on first version you have at least the first debug message
```
Starting application
Config version: 0
Clearing Global Config
Screen init done
```

Have yout tried to rollback the code like dexribed here https://github.com/Smanar/CYD-Domoticz-Remote/issues/4#issuecomment-1992110622
You need to have at least the same result than before ?

franki29, 2024-03-15:
Here is what I am using, still no serial out

[env:esp32-4848S040]
board = esp32-s3-devkitc-1
framework = arduino
;extends = esp32
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.10.03/platform-espressif32.zip
;platform = espressif32

lib_deps =
	${env.lib_deps}
	moononournation/GFX Library for Arduino@1.4.0
	;https://github.com/PaulStoffregen/XPT2046_Touchscreen.git
	Wire
	git+https://github.com/netwizeBE/arduino-goodix.git

board_build.arduino.memory_type = qio_opi
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.mcu = esp32s3

board_build.f_cpu = 240000000L  ; Set main frequency to 240MHz
board_build.partitions = min_spiffs.csv  ; Use partitions that allocate 512KB for SPIRAM

board_build.psram = enable  ; Enable PSRAM
board_build.psram_config = ${platform}_${board} 8MB  ; Set PSRAM size to 8MB

board_upload.flash_size = 16MB

build_flags =
	${env.build_flags}
	#-DTFT_ESPI=1
	#-DLOVYANGFX=1
	-DARDUINO_GFX
	;-DTOUCH_911
	-DTOUCH_DISABLED
	-DESP32_4848S040
	-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue  ; Enable PSRAM in code
    -DTOUCH_SDA=19
    -DTOUCH_SCL=45
    -DTOUCH_RST=-1
    -DTOUCH_IRQ=-1
    -DI2C_TOUCH_FREQUENCY=400000
    -DI2C_TOUCH_ADDRESS=0x5D ; or 0x14
    -DTFT_WIDTH=480
    -DTFT_HEIGHT=480
	-DTOTAL_ICONX=3 # How many icon widht
	-DTOTAL_ICONY=3 # How many icon Heigh
	-DDEVICE_SIZE=1 # Device display size

Smanar, 2024-03-15:
Strange this is the code that have started https://github.com/Smanar/CYD-Domoticz-Remote/commit/f8ac9d1af0e4a5eae9f507da33abf1ea015bb107

```
[env:esp32-4848S040]
board = esp32-s3-devkitc-1
framework = arduino
;extends = esp32
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.10.03/platform-espressif32.zip

lib_deps =
	${env.lib_deps}
	moononournation/GFX Library for Arduino@1.4.0
	;;https://github.com/PaulStoffregen/XPT2046_Touchscreen.git
	Wire
	git+https://github.com/netwizeBE/arduino-goodix.git

board_build.arduino.memory_type = qio_opi
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_upload.flash_size = 16MB
board_build.mcu = esp32s3

board_build.f_cpu = 240000000L  ; Set main frequency to 240MHz
board_build.partitions = min_spiffs.csv  ; Use partitions that allocate 512KB for SPIRAM

board_build.psram = enable  ; Enable PSRAM
board_build.psram_config = ${platform}_${board} 8MB  ; Set PSRAM size to 8MB

build_flags =
	${env.build_flags}
	#-DTFT_ESPI=1
	#-DLOVYANGFX=1
	-DARDUINO_GFX
	-DTOUCH_911
	-DESP32_4848S040
	-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue  ; Enable PSRAM in code
    -DTFT_WIDTH=480
    -DTFT_HEIGHT=480
	-DTOTAL_ICONX=3 # How many icon widht
	-DTOTAL_ICONY=3 # How many icon Heigh
	-DDEVICE_SIZE=1 # Device display size
```

It's the same, you are using platformio application ? If yes try to clean the project, the garbage icon, close all task, recompile and after try to use the serial monitor.

Here https://github.com/Smanar/CYD-Domoticz-Remote/tree/f8ac9d1af0e4a5eae9f507da33abf1ea015bb107

You have the complete code at this step, but for me it can't be from the code.
There is some change on the code, but nothing that can prevent the code totaly starting.

franki29, 2024-03-16:
Hi, i even change from Windows to Linux, same problem.

Using your older Version, it is stopping at the screen settings and reboots:
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8
Starting application
Config version: 0
Clearing Global Config
Screen init done

Smanar, 2024-03-16:
> Using your older Version, it is stopping at the screen settings and reboots

Ok, so as it's better with this version, we will start by this one, step by step.

I have a created a new branch, just for your device called "franki".
It's based on the same branch, for the moment I have just added some more debug line https://github.com/Smanar/CYD-Domoticz-Remote/commit/c5c9b70f4a23564f45dc9b49afa9c33809d1a0a5

So this can prevent the device starting, I hope some more debug line after 

```
Starting application
Config version: 0
Clearing Global Config
Screen init done
```

franki29, 2024-03-16:
Guru Meditation Error: Core  1 panic'ed (Double exception). 

Core  1 register dump:
PC      : 0x403743c0  PS      : 0x00060d36  A0      : 0x8a4e51f4  A1      : 0x600ffff4  
A2      : 0x00060d36  A3      : 0x00040026  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x0000ffff  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0xfa000000  A11     : 0x00000194  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x01000000  A15     : 0x00ffffff  SAR     : 0x0000000a  EXCCAUSE: 0x00000002  
EXCVADDR: 0x00000000  LBEG    : 0x4201d664  LEND    : 0x4201d686  LCOUNT  : 0x00000000  

Backtrace: 0x403743bd:0x600ffff4 0x4a4e51f1:0x601000b4 |<-CORRUPTED

Core  0 register dump:
PC      : 0x420acc16  PS      : 0x00060034  A0      : 0x82040731  A1      : 0x3fcf40d0  
A2      : 0x00000000  A3      : 0x420a1778  A4      : 0x00060120  A5      : 0x80000000  
A6      : 0x00ffffff  A7      : 0x3fcadaa4  A8      : 0x820402a6  A9      : 0x3fcf40a0  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x8037ca88  A13     : 0x3fcf3fd0  
A14     : 0x00060023  A15     : 0x00000003  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x420acc13:0x3fcf40d0 0x4204072e:0x3fcf40f0 0x420af568:0x3fcf4110

ELF file SHA256: 0000000000000000

Rebooting...
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42040d9f
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8
Starting application
Config version: 0
Clearing Global Config
Screen init done
Debug 1

Smanar, 2024-03-16:
```
Starting application
Config version: 0
Clearing Global Config
Screen init done
Debug 1
```

So it crash at wifi initialisation ?
Your device have wifi  ?

franki29, 2024-03-16:
load:0x403cc700,len:0x28cc
entry 0x403c98b8
Starting application
Config version: 0
Clearing Global Config
Screen init done
Debug 1

Debug 7

Smanar, 2024-03-18:
Oups, sorry have missed your answer.
I have pushed last change on branch "test", sorry but too hard for me to work on 2 branch in same time with the same editor.

So on the last version:
- Use a new code, take from here https://homeding.github.io/boards/esp32s3/panel-4848S040.htm
- Some minor chnage on screen drivers
- More logs again.

On your last tests the code have crashed on fonction wifi_init_inner()
But this fonction try to set the wifi and it's the first use of lvlg, so the problem can come from both.

franki29, 2024-03-19:
Hi, i tested the new test branch.
Now there is no reset anymore, but screen keeps dark with backlight on.
Here is the serial output:

rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8
Starting application
Config version: 0
Clearing Global Config

Smanar, 2024-03-19:
No reset, but not sure it's better, now the code is freezed ealier....

Ok so rollback on branch "franki" again, sorry, but I m not able to find a code on github certified working, some are using old versions, some other are "working" but according other post the code is not working.

So I have made a minimal code https://github.com/Smanar/CYD-Domoticz-Remote/blob/franki/src/main.cpp

You have only 1 file, it don't use lvlg yet, it's just the base.
If this one nor working no need the rest.

Something you can try if it still not working, replace `st7701_type1_init_operations` used by defaut by the lib by  `st7701_4848S040_init_operations` modified version from OpenHASP.

As this code is minimal, I will be able to ask help on the lib github.

franki29, 2024-03-19:
Arduino_DataBus* bus            = new Arduino_SWSPI(TFT_DC, TFT_CS, TFT_SCLK, TFT_MOSI, TFT_MISO); is working and Hello Worlds are shown on the display

Smanar, 2024-03-19:
Nice, have updated the "franki" branch, now the code will display a text again but using lvlg now.

franki29, 2024-03-20:
st7701_4848S040_init_operations

but the lvlg demo is not running. Even I change st7701_4848S040_init_operations or Arduino_DataBus* bus = new Arduino_SWSPI(TFT_DC, TFT_CS, TFT_SCLK, TFT_MOSI, TFT_MISO);

franki29, 2024-03-20:
Nothing on the serial output

Smanar, 2024-03-20:
So the problem is definitively with lvlg.
Have send a new version, but it's not logic you haven't serial output at all .... At least the debug line before the lvlg stuff.

franki29, 2024-03-21:
The new one is working wih Arduino_DataBus* bus = new Arduino_SWSPI(TFT_DC, TFT_CS, TFT_SCLK, TFT_MOSI, TFT_MISO);

Smanar, 2024-03-21:
Seriously ?
It work ?
You have a message "Hello arduino" and a version number ?

franki29, 2024-03-22:
[image]

Smanar, 2024-03-22:
Have pushed a new version.
Now it will print a text with color, "red" in red, "blue" in blue and ect, to test color.
Have remove unused code
Have re-used the old lv_conf.h file, its the one I m using for the main project.

franki29, 2024-03-22:
The new version only works when I change 
// 9-bit mode SPI
Arduino_DataBus* bus = new Arduino_SWSPI(TFT_DC, TFT_CS, TFT_SCLK, TFT_MOSI, TFT_MISO);

Also no colors a shown. Same background color like in my picture and white text with the color definitions

franki29, 2024-03-23:
Hi, Blue is yellow, green is purple RED something in direction white, but it is so small that I can not really see.
Background is still Orange.

Smanar, 2024-03-23:
Ok so all is "normal", you have the same issue than here https://github.com/HASwitchPlate/openHASP/issues/603#issuecomment-1874422572

So can try with replacing `st7701_type1_init_operations` used by defaut by the lib by` st7701_4848S040_init_operations` modified version from OpenHASP. 

So have pushed another version, inside.
- replaced the init operation, to see if color are fine now.
- line 250, replaced `if (!gfx->begin(11000000))` by `if (!gfx->begin())` because I think it will impact performance and not working mode.
- More suspicious code, something complex about buffering, you can disable the code by replacing `#if 0` by `#if 1` on line 271 if I have broked the code again.

franki29, 2024-03-23:
Yes, now it is OK. White background and text in the color they should be

Smanar, 2024-03-23:
Nice, so next try will be the last.
I have just cleaned the code.

line 217/218
```
const size_t VDBsize = LV_VDB_SIZE / sizeof(lv_color_t);
//const size_t VDBsize = 480*480;
```

If it don't work, can you try with the second line (switch comment) and with the same value by / 8 (so const size_t VDBsize = 480*480 / 8;)

After that 2 choice, adapt the official code or test the touchpad on the "franki" branch (faster to compile)

franki29, 2024-03-23:
Still OK and running without changing the second line.

I think we can try with the official code

Smanar, 2024-03-23:
Done.
You can try again the "test" branch.
Have enabled the untested touchpad code, so you can be blocked at calibration screen, but you will see something at screen, and I hope no more crash.

franki29, 2024-03-23:
Not sure what I am doing wrong, but now I get compiling errors:

In file included from .pio/libdeps/esp32-4848S040/WebSockets/src/WebSocketsClient.h:28,
                 from src/core/ip_engine.cpp:2:
.pio/libdeps/esp32-4848S040/WebSockets/src/WebSockets.h:206:10: fatal error: WiFiClientSecure.h: No such file or directory

**************************************************************************
* Looking for WiFiClientSecure.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClientSecure.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClientSecure.h
*
**************************************************************************

  206 | #include <WiFiClientSecure.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/esp32-4848S040/src/core/ip_engine.cpp.o] Error 1
In file included from .pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/databus/Arduino_ESP32SPI.h:8,
                 from .pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/Arduino_GFX_Library.h:24,
                 from include/../drivers/esp32-4848S040.h:2,
                 from src/core/screen_driver.cpp:67:
/home/frank/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32s3/include/driver/deprecated/driver/periph_ctrl.h:7:2: warning: #warning driver/periph_ctrl.h header is no longer used, and will be removed in future versions. [-Wcpp]
    7 | #warning driver/periph_ctrl.h header is no longer used, and will be removed in future versions.
      |  ^~~~~~~
In file included from .pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/Arduino_GFX_Library.h:4:
.pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/Arduino_DataBus.h:182:13: error: 'i80_device_list' has not been declared
  182 |   LIST_HEAD(i80_device_list, lcd_panel_io_i80_t)
      |             ^~~~~~~~~~~~~~~
.pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/Arduino_DataBus.h:182:3: error: ISO C++ forbids declaration of 'LIST_HEAD' with no type [-fpermissive]
  182 |   LIST_HEAD(i80_device_list, lcd_panel_io_i80_t)
      |   ^~~~~~~~~
.pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/Arduino_DataBus.h:182:48: error: expected ';' at end of member declaration
  182 |   LIST_HEAD(i80_device_list, lcd_panel_io_i80_t)
      |                                                ^
      |                                                 ;
.pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/Arduino_DataBus.h:183:3: error: 'device_list' does not name a type
  183 |   device_list; // Head of i80 device list
      |   ^~~~~~~~~~~
.pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/Arduino_DataBus.h:217:3: error: ISO C++ forbids declaration of 'LIST_ENTRY' with no type [-fpermissive]
  217 |   LIST_ENTRY(lcd_panel_io_i80_t)
      |   ^~~~~~~~~~
.pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/Arduino_DataBus.h:217:32: error: expected ';' at end of member declaration
  217 |   LIST_ENTRY(lcd_panel_io_i80_t)
      |                                ^
      |                                 ;
.pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/Arduino_DataBus.h:218:3: error: 'device_list_entry' does not name a type
  218 |   device_list_entry; // Entry of i80 device list
      |   ^~~~~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.h:18,
                 from .pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/Arduino_GFX_Library.h:19:
/home/frank/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32s3/include/hal/esp32s3/include/hal/lcd_ll.h: In function 'void lcd_ll_enable_interrupt(lcd_cam_dev_t*, uint32_t, bool)':
/home/frank/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32s3/include/hal/esp32s3/include/hal/lcd_ll.h:557:33: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  557 |         dev->lc_dma_int_ena.val |= mask & 0x03;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/frank/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32s3/include/hal/esp32s3/include/hal/lcd_ll.h:559:33: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  559 |         dev->lc_dma_int_ena.val &= ~(mask & 0x03);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
.pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.h: At global scope:
.pio/libdeps/esp32-4848S040/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.h:43:3: error: 'esp_lcd_rgb_panel_frame_trans_done_cb_t' does not name a type; did you mean 'esp_lcd_panel_io_color_trans_done_cb_t'?
   43 |   esp_lcd_rgb_panel_frame_trans_done_cb_t on_frame_trans_done; // Callback, invoked after frame trans done
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   esp_lcd_panel_io_color_trans_done_cb_t
*** [.pio/build/esp32-4848S040/src/core/screen_driver.cpp.o] Error 1

Smanar, 2024-03-23:
Ha ? Something new ....
Have you tried after deleting the folder ".pio" (created automatically in the project folder)

franki29, 2024-03-24:
So I had to change 
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.10.03/platform-espressif32.zip
;platform = espressif32

for compiling, but than it did not run. After disabling the touch, I seem to work as calibration screen was shown.

So the g911 touch driver is not working yet

Smanar, 2024-03-24:
Ha yes, right, it was like that on branch "franki", but strange it compile on my side ! Perhaps something happen is the device is plugged ?

I m looking the touch code, the only blocking part for me is on line 183 > 238, so have added some debug lines on new version (5mn ago), can you tell me if you have some debug output with it ?

franki29, 2024-03-24:
No debug, just  rebooting all the tine:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8

Smanar, 2024-03-24:
And without log again ? WTF I realy hate your device ...

So have adeed in code 
```
#if DEBUG1
#if DEBUG2
#if DEBUG3
```

Those line disable all this code part, you can enable part one by one with adding a "!" like `#if !DEBUG3`

By defaut the code have all the code disabled, I have just let lines

```
#include <Wire.h>
#include "Goodix.h"

static Goodix touch = Goodix();
```

But seriously if just an import can crash your device, it's realy a problem for devs.

I have found another lib for your touchpad but not used so much https://github.com/TAMCTec/gt911-arduino

franki29, 2024-03-24:
When I make !DEBUG1 than it does not crash but also it does not print "Init 911 driver" nor 4444444444.

It directly print Starting calibration

franki29, 2024-03-24:
!DEBUG2 is crashing

Smanar, 2024-03-24:
Haa .....
I remember some issue on eps32, this can happen is the serial is not ready, have added a 500ms delay after the serial initialisation, your device is probably to fast.

So new version online, with same DEBUGX to disable.
Have just added some delay, because all the code is fine for me, I can understand it don't work, but not it can crash ....

franki29, 2024-03-24:
Starting application
Config version: 0
Clearing Global Config

With no debug on

franki29, 2024-03-24:
So it is stopping after clearing config

Smanar, 2024-03-24:
So there is no more crash, but not working ?

Smanar, 2024-03-24:
Ha fuck I have understand why you haven't other log, and why it crash before the logs.

Smanar, 2024-03-24:
Ok so this time was my fault.
Have re write the touch code for all device, It will still crashing I think, but this time we will have logs.

Can you tell me how it s with debug disabled ? all are enabled by defaut, so no crash at first try.

franki29, 2024-03-25:
Starting application
Config version: 3
Init 911 driver

44444444444

Driver GT911 started: (480x480)
888888888
Check ACK on addr request on 0x
5D
: SUCCESS
9999999999
Starting calibration
Calibration coef: inf , inf
Calibration offset: nan , nan
Display buffer size: 24576 bytes
Screen init done
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected

franki29, 2024-03-25:
So when I try to select the network, nothing is happening

Smanar, 2024-03-25:
Ok so it mean all was my fault, sorry, was not a problem with your device (at least for the touchpad).
So I think this lib can work now, I have updated the code again 

```
Starting calibration
Calibration coef: inf , inf
Calibration offset: nan , nan
```

If all is fine you will have value now here, else I m wrong again on my side. And thx for your patience.

franki29, 2024-03-26:
Now calibartion is not shown.

Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb90
load:0x403cc700,len:0x28cc
entry 0x403c98b8
Starting application
Config version: 3
Init 911 driver

Driver GT911 started: (480x480)
Check ACK on addr request on 0x
5D
: SUCCESS
Display buffer size: 24576 bytes
Screen init done
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected
WiFi Status: Disconnected

Smanar, 2024-03-26:
Yes it's normal, it's because you had bad value but the calibration was finished, so the device don't ask for it again.

`	-DFORCECALIBRATE=true # Reset the calibration at every restart, used to reset it after a config change.`

If you want I can make a special branch in "franki" with code specific to your device, but for me we are realy close the end.

franki29, 2024-03-26:
: SUCCESS
Starting calibration
Calibration coef: 0.9 , 0.9
Calibration offset: 13.5 , 6.8
Display buffer size: 24576 bytes
Screen init done
WiFi Status: Disconnected
WiFi Status: Disconnected

Smanar, 2024-03-26:
Value don't look so bad.
There is an hidden tool in the code in the screen driver.cpp file
```

#if 0
//Tool to test the touchpad, need to be disabled, only for debug purpose
while (true)
{
    p = touchscreen.getPoint();
    p.x = round((p.x * global_config.screenCalXMult) + global_config.screenCalXOffset);
    p.y = round((p.y * global_config.screenCalYMult) + global_config.screenCalYOffset);

    tft.drawFastHLine(p.x,p.y,2,TFT_RED);
}
#endif
```

Just replace `#if 0` by `#if 1`
It will block the code, but just after the calibration, you will be able to "color a pixel" with pressing the display, usefull to see if the screen is not reversed, or a bad scale.

Smanar, 2024-03-26:
Have upodated the code again, you can let 
`-DFORCECALIBRATE=false`

for me your last try was fine.

[9 more replies]

closed 2024-03-26 by franki29