Smanar/CYD-Domoticz-Remote #5: Text in light mode not visible

franki29, 2024-03-26:
Hi me again :-)

If I switch to light, mode most of the text is not visible.

Smanar, 2024-03-27:
Hello again ^^.
Have updated the theme for the light mode, but it's not perfect. There is white in color used for button, and the theme use it too, so in some situation there is 2 things on the same color.
Problem, I don't found how to change the white color for button and If I use a light grey, the white color is not visible enought.

franki29, 2024-03-28:
Additional you clean the test code for the screen driver, but now it can not be compiled anymore if you select the  default_envs = esp32-4848S040

franki29, 2024-03-28:
PS Light mode works fine for my device. But I had to reboot to make it work

closed 2024-03-28 by franki29

Smanar, 2024-03-28:
Additional you clean the test code for the screen driver, but now it can not be compiled anymore if you select the default_envs = esp32-4848S040

Arf, thx for information, have updated Stable branch too.

And I think now it work without restart.