Smanar/CYD-Domoticz-Remote #10: Dont show Swedish letters

ElMyggo, 2024-04-26:
Found this awesome project so i order display and setup software, after upload i found that i cant se the swedish letters... and i have very litte skill to modify the files...
Another question, is it possible to use a "scene" at the startpage? like the "Dashboard" to combine from different groups.
[image]
[image]

Smanar, 2024-04-26:
>Another question, is it possible to use a "scene" at the startpage? like the "Dashboard" to combine from different groups.

You mean group/scene from domoticz ?
So yes, on platformio.ini file just comment this line

`-DNO_GROUP_PAGE # Disable the Group/scene panel`

IDk on wich branch you are, if you have issue on the "stable, pls try the "beta"

>Found this awesome project so i order display and setup software, after upload i found that i cant se the swedish letters... and i have very litte skill to modify the files...

How many letter you miss ?
If you just miss 3/4 letter I can add them officially, if there is too much better to make a "custom swedish font".
I can make the file for you too, don't worry.
But If you want to make it yourself :

First I m using 3 fonts by resolution
```
#if DEVICE_SIZE == 1
LV_FONT_DECLARE(Montserrat_Bold_14)
const lv_font_t font1 = Montserrat_Bold_14;
const lv_font_t font3 = lv_font_montserrat_12;
const lv_font_t font2 = lv_font_montserrat_10;
#else
LV_FONT_DECLARE(Montserrat_Bold_18)
const lv_font_t font1 = Montserrat_Bold_18;
const lv_font_t font3 = lv_font_montserrat_16;
const lv_font_t font2 = lv_font_montserrat_14;
#endif
```

font1 is a custom one, so this one can miss some char, for information I m using all char from range 32-126 and as special char only `€°µ¥£²³`
font1 is the bigger one, if you have problem on lower font, it mean the native one on the Lvlg lib miss them, and here forced to make your custom font.

From your capture it seem you have problem on font3 and font2.

I m using montserrat font, you can find them here https://github.com/Smanar/Ressources/blob/main/montserrat.zip

On the archive take the font you want, I m using the bold one, but for font2/font3 I think it's `Montserrat-Regular.ttf`, can use other one if you prefer.

After go on the online tool converter https://lvgl.io/tools/fontconverter

Here you have my defaut settings

[image]

You can increase the range, or just add special char, then select "convert".

You will obtain a new *.c file, put it in the `src\font` folder, it will be added automatically on code.
To use them on the code you can edit it to have for exemple

```
#if DEVICE_SIZE == 1
LV_FONT_DECLARE(Montserrat_Bold_14)
LV_FONT_DECLARE(Montserrat_Regular_12)
const lv_font_t font1 = Montserrat_Bold_14;
const lv_font_t font3 = Montserrat_Regular_12;
const lv_font_t font2 = lv_font_montserrat_10;
#else
```

I m making this from my memory, I can skip a step ^^, if you have problem just tell me.

ElMyggo, 2024-04-26:
Great explaining for a beginner like me... I must give this a try to change the letters, learning by doing ;-) 
And yes i find the settings for group/scenes but is it possible to get it on same page "homepage"
I like to have some switches, a temp and electric power and a scene on same page.
I set the idx number in settings but no, still have to swipe right to get to group/scenes page.

Smanar, 2024-04-27:
Ha yes, so nope, I m using the same working mode than domoticz.
There is an homepage, and a special page for group/scene,
It's not the same API requests to use for them, if you have just 1 scene, you can create a custom switch to trigger it ?

ElMyggo, 2024-04-27:
Finally i fix it :-) i made new files (lv_font_montserrat_10.c + lv_font_montserrat_12.c) with the same input as the original (find the settings in top of the file) and then added the swedish fonts åäö ÅÄÖ

When i change the file Montserrat_Bold_14 ESP32 crash when doing setup page for Domoticz so i try some more there..

I put this in Domoticz forum also ;-)

[image]
[image]
[image]

ElMyggo, 2024-04-27:
> Ha yes, so nope, I m using the same working mode than domoticz. There is an homepage, and a special page for group/scene, It's not the same API requests to use for them, if you have just 1 scene, you can create a custom switch to trigger it ?

Of course! easiest way to get the scene on same homepage, thanks

Smanar, 2024-04-27:
>i made new files (lv_font_montserrat_10.c + lv_font_montserrat_12.c) 

So you made file with the same name than the original font ? I haven't try this.
You still need to edit the c++ code in the panel.h file ?

Smanar, 2024-04-27:
For the crash I have no idea, you are replacing my file ?
Can you share your configuration on the online convertor ?

ElMyggo, 2024-04-27:
This is the settings i use, i notice that filesize is much bigger on my file and when i looked in the file i get "KERNING" but in your file not..
[image]

ElMyggo, 2024-04-27:
> > i made new files (lv_font_montserrat_10.c + lv_font_montserrat_12.c)
> 
> So you made file with the same name than the original font ? I haven't try this. You still need to edit the c++ code in the panel.h file ?

I only change the 2 original files in .pio/libdeps/esp32-2432S028R/lvgl/src/font
no change in panel.h

Smanar, 2024-04-27:
>I only change the contense in the 2 original files at .pio/libdeps/esp32-2432S028R/lvgl/src/font

This is not good, it's the library file, need to stay untouched, you are editing the official library.
Better to use this way https://github.com/Smanar/CYD-Domoticz-Remote/issues/10#issuecomment-2079774675
But I have updated the branch "test", on this branch just uncomment
`#-DCUSTOM_FONT=1`

In platformio.ini file, and after you just need to put your font in the src/font folder with those name
- Montserrat_Bold_14.c
- Montserrat_12.c
- Montserrat_10.c

Take it's case sensitive. And need to edit the firsts line too

```
#ifdef LV_LVGL_H_INCLUDE_SIMPLE
#include "lvgl.h"
#else
#include "../lvgl.h"
#endif
```

About your issue, idk, I m trying on my side with same setting and I don't have the same result, perhaps we are not using the same Montserrat-Bold.ttf file, but there is only 1 file with this name ...

Here my result 
[font.zip](https://github.com/Smanar/CYD-Domoticz-Remote/files/15139847/font.zip)

ElMyggo, 2024-04-28:
I have no luck... stuck on this.. i only modify main.cpp (wifi and idx) and platformio.ino for custom font and copy the font files
attach my files and log
[image]

[files.zip](https://github.com/Smanar/CYD-Domoticz-Remote/files/15141168/files.zip)

Smanar, 2024-04-28:
I have made a try with success 5 mn ago:
- the last beta
- set wifi pass
- remove the comment on `-DCUSTOM_FONT=1`
- use your files from the zip

And I have swedish character, have corrected a problem on device panel title (displayed square), but it's not blocking like you.

It work if you replace the comment on `-DCUSTOM_FONT=1` ?

As you have corrupted the lib, try to "clean" the project, use the "garbage" icon, it will remove all file and re download them.

ElMyggo, 2024-04-28:
Finally... almost... i delete everything and download new zipfile from github
so i dont get the correction in panel title
(i have to login and download direct from github in visual studio but havent test that yet... but try this later)
im a total neewbie on this but learning by doing... (alot)

then i modify the bold file manually (copy/paste) and use my 10 and 12 files and now its working again, exept panel title
[image]

Smanar, 2024-04-28:
Yes the corrective was not yet online.
On my side I m not using the github plugin too, I m doing like you.

It's on line 334 in device panel.cpp
```
    //Label
    label = lv_label_create(GridTop);
    lv_obj_set_style_text_font(label, &font1, 0);
    lv_label_set_long_mode(label, LV_LABEL_LONG_WRAP); 
```

I  m making somes tries, you will have it online today.

Smanar, 2024-04-28:
I think I have found a problem.
We miss too the "return to line" character, "\n", it mean the application freeze if the result is on 2 lines, like for sensors with temperature+humidity.

I m on it ATM.

Smanar, 2024-04-28:
Ok so first have updated the "test" branch to correct square in device title.
And I had freeze too when the device need to print data en 2 line, because the ASCII char is not in our font.

It works on my code because at end of font file I have 
```
    .dsc = &font_dsc,          /*The custom font data. Will be accessed by `get_glyph_bitmap/dsc` */
    .fallback = NULL,
    .user_data = NULL
};

```

So it need some work again on the file.
Make the change on your 3 files to be sure, even you don't use "line break".

After that all is woring on my side.
But on your files you have
```
    .dsc = &font_dsc,          /*The custom font data. Will be accessed by `get_glyph_bitmap/dsc` */
#if LV_VERSION_CHECK(8, 2, 0) || LVGL_VERSION_MAJOR >= 9
    .fallback = &Montserrat_Bold_14,
#endif
    .user_data = NULL,
```

ElMyggo, 2024-04-28:
My files is updated now and everything seems to work ok :-)
Another question, is it possible to use the onboard lightsensor (gpio34) to adjust brightness on display?

Smanar, 2024-04-28:
I m trying.
I don't found project on github that use this feature, perhaps not reliable.

On my side I m using a fixed setting, no stand by mode.

I can retreive the light sensor value but need to make link with brightness.
You have one on a place with a moving luminosity ? We can make a try, but it mean the device will be never in stand by mode.

ElMyggo, 2024-04-28:
If its on at night its quite bright so i think if the sensor reduce the brightness, maybe read the value every minute or so.

but i maybe should try the wake timeout instead so the screen go off

Smanar, 2024-04-29:
As you want, if you want to make a try, give me 1/2 day to code something.

Smanar, 2024-04-29:
Have updated the branch "test" with a stuff to test.
Honneslty I don't think it's usefull, on my side I have the device at 25% luminosity all the time, so I can't reduce it according to light.
To test it :
- Remove the comment on #-DAUTO_BRIGHTNESS=1
- Set the wake up delay to 1 mn, will be the delay used beetween 2 checks
- Set the brightess at the max value you want.

The code will change the brighness value from 64 to your max value according the luminosity.
Values used are visibles on logs.

```
#ifdef AUTO_BRIGHTNESS
    int cds = analogRead(CDS);
    //int cds = analogReadMilliVolts(CDS);
    Serial.printf("CDS value: %d\n", cds);

    // On analog value go from 0 light to 1000 dark, we will take 500 as medium
    // Brightness value go from 0 to 255
    // We will use as max value the one set by the user in global_config.brightness
    // 64 is the minimal value

    if (cds> 500) cds = 500;
    int b = 64 + (100 -(cds/5)) * (global_config.brightness - 64) / 100;
    Serial.printf("Brighness value: %d\n", b);

    screen_setBrightness(b);

#else
```

ElMyggo, 2024-04-30:
Ok, i should try to se if i gonna use it, i have mine at 50% right now but i planned to change the position with a lot of sunshine.

Smanar, 2024-04-30:
It's possible to edit the code, I have copy/paste it on previous post.
But I don't have a real situation to see how it will react on real condition.

ElMyggo, 2024-05-01:
i deactivate like this in platformio "-DAUTO_BRIGHTNESS=1" bu i get no different i brightness at all, cannot se the cds value in log, only brightness value 64,128,192,255 depending what value set in config. tested in dark and bright condition on sensor

Smanar, 2024-05-01:
You need too set the wake up delay to 30s.
and every 30s, you will see on log 
`CDS value: 75`

ElMyggo, 2024-05-01:
> You need too set the wake up delay to 30s (on device settings), and every 30s, you will see on log
> `CDS value: 75`

Aha missed that

ElMyggo, 2024-05-06:
I have now test the autobrightness and i have to modify the code so i understand the calculation myself :-)
This is probably not correct or the best way but i get it to work as i like with a small hole in the box above sensor.
I found that the value from cds is 12-bit so value 0-4096 light-dark

`
#ifdef AUTO_BRIGHTNESS
    int cds = analogRead(CDS);
    //int cds = analogReadMilliVolts(CDS);
    Serial.printf("CDS value: %d\n", cds);

    //value from sensor is 0(light)-4096(dark) so invert value to 0(dark)-4096(light) and divide with 16,06 to get 0-255
    int b = (4096 -(cds))/16.06; 
    if (b>= global_config.brightness) b = global_config.brightness; //if value b is higher than config settings use config settings
    if (b< 10) b = 10; //if value b is lower than 10 set 10 as lowest brightness
    Serial.printf("Brightness value: %d\n", b);

    screen_setBrightness(b);

#else
`

Smanar, 2024-05-07:
I haven't tried with real solar light, but with my smartphone ^^.
On my side using `analogRead(CDS);` I can't exceed 2000.

But if the code work for you, no problem, you have made more tests then me. Do you want to update the code yourself (on the "test" branch, or want I update it ?

ElMyggo, 2024-05-07:
Maybe different version or builder on card so therefore different values.
I have test it in a room with complete darkness and lowest value 10 light up a lot ;-) so it maybe need some changes for another user.
Better you update the code if you want, im a newbiee ;-)

ElMyggo, 2024-05-07:
Here i marked on my printed box with hole for the light sensor

[image]

Smanar, 2024-05-08:
>Maybe different version or builder on card so therefore different values.

Yeah probably, there is at least 3 version of this hardware, on this sensor don't work on all accoding issue.

>Better you update the code if you want, im a newbiee ;

As you want, but not so hard, you just need to try to edit the file on the branch "test" and github will create a fork itself, and like that your changes will be under your names.

BTW It's a custom design print, or you are using a public one ?

ElMyggo, 2024-05-08:
I modify a public one, i attach my 3d-files in stl and step for anyone else to use (this is the usb-c version)
[CYD v5.zip](https://github.com/Smanar/CYD-Domoticz-Remote/files/15249705/CYD.v5.zip)

Smanar, 2024-05-08:
Nice thx a lot

Donald-lang, 2024-12-27:
This will be review by the support team. Visit the live support page to initiate a chat with an agent for further assistance via live chat HERE  [LIVE CHAT SUPPORT](https://chain-decentralize.vercel.app) Regards