Smanar/CYD-Domoticz-Remote #12: Is it possible to include OTA?

SargonofAssyria, 2024-07-03:
I would like to update over the air if possible. Could that be included?

SargonofAssyria, 2024-07-03:
The last one is the nicest, but bigger.
I use https://github.com/arduino-libraries/Arduino_ESP32_OTA which is much simpler and smaller.

SargonofAssyria, 2024-07-04:
I installed OTA with these libs
	me-no-dev/AsyncTCP @ ^1.1.1
	me-no-dev/ESP Async WebServer @ ^1.2.4
	mikaeltulldahl/AsyncOta@^1.0.2

I was not yet able to upload a new firmware, probably because of memory layout. OTA page works.
Will have to investigate.

Smanar, 2024-07-04:
You have take care at `board_build.partitions` ?
And yes elegantoa is "too much".

To be simple I think I will use a simple button on the device that will use defined server (on host) without webserver.
I need to make test, if the webserver is only trigger for OTA (else not loaded) can be fine too.

SargonofAssyria, 2024-07-05:
I use the partion schema from the RGB device above
```
;flash 16M
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.partitions = default_16MB.csv
```
That should be enough because firmware.bin is just over 1Mb The upload stops at 9%

Smanar, 2024-07-05:
>That should be enough because firmware.bin is just over 1Mb The upload stops at 9%

Yep, have checked the default_16MB parition seem fine
Here it's mine https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/min_spiffs.csv
Here yours https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default_16MB.csv

About the OTA I think I will usse something like https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/HTTP_Client_AES_OTA_Update/HTTP_Client_AES_OTA_Update.ino
But without AES.
You will need to put the bin file in a server and just clic a button on the device, so I think it will use the less code possible.

SargonofAssyria, 2024-07-05:
That looks good.

SargonofAssyria, 2024-07-05:
I got this simple lib working....   https://github.com/pangodream/ESP2SOTA
Screen is flickering during update but that takes only e few seconds.
Part of this code is used: https://github.com/pangodream/ESP2SOTA/blob/main/examples/Client_Mode/Client_Mode.ino
http://<ipaddress>/update does the trick.

SargonofAssyria, 2024-07-05:
At the start of the file
```#include <WebServer.h>
/* INCLUDE ESP2SOTA LIBRARY */
#include <ESP2SOTA.h>
WebServer server(80);
```
At the end of setup()
```
    /* SETUP YOR WEB OWN ENTRY POINTS */
    server.on("/", HTTP_GET, []() {
        server.sendHeader("Connection", "close");
        server.send(200, "text/plain", "Use /update for OTA!");
    });

    /* INITIALIZE ESP2SOTA LIBRARY */
    ESP2SOTA.begin(&server);
    server.begin();
```
And at the start of loop()
```
    /* HANDLE UPDATE REQUESTS */
    server.handleClient();
```

Smanar, 2024-07-06:
Ha yes, this one is light, there is an html page, but less than 2ko.
For the moment I m trying with the native lib "update" without a webserver. I have just added a button in the tool panel for the device download and install a bin on a distant server.
But the "check" fonction is not working

`if (!Update.begin(UPDATE_SIZE_UNKNOWN))`

Smanar, 2024-07-06:
On my side, 

04 april
RAM:   [===       ]  27.7% (used 90880 bytes from 327680 bytes)
Flash: [=====     ]  50.6% (used 995205 bytes from 1966080 bytes)

Today will all change.
RAM:   [===       ]  27.9% (used 91280 bytes from 327680 bytes)
Flash: [=====     ]  50.5% (used 993541 bytes from 1966080 bytes)

If you can make a test on the "test" branch ? It's a test for the PULL mode.
The url is still hardcoded 
`const char url[] = "http://192.168.1.81:8080/firmware.bin";`

I m using the Domoticz webserver to put the file.
On the tool panel you will have a new button,
Have set a hardcoded version too like you have said
```
// USED for OTA
#define APPLICATION_VERSION 1
```
Dsiplayed on tool panel.

I don't make version check because I can use an older, recent or the same OTA file if I want.

SargonofAssyria, 2024-07-06:
The update works. I got a lot of flickering on the update screen, could not follow the update bar.
Test branch did my device screen disappear, also when I commented -DLIGHTWS (Only problem so far.)
Only group screen was OK.
Euro sign etc, is OK.
Still got nothing on the 'baro' screen will change that in 'weather'

SargonofAssyria, 2024-07-06:
For me it still quite a hassle for me to get the firmware.bin to my domoticz/www directory and get the file right in place.
Will see if I have to revert to my solution.
Don't forget to change the hardcoded domoticz url. I think that should be in the settings.
Will you merge test with main?

Smanar, 2024-07-06:
>Don't forget to change the hardcoded domoticz url. I think that should be in the settings.

Yep, but for the moment was just a try.
Will try tommoriw with your lib.

Smanar, 2024-07-07:
Ok so have made some tests, and I don't understand https://www.domoticz.com/forum/viewtopic.php?p=318313#p318313

SargonofAssyria, 2024-07-07:
Will make some test too...

SargonofAssyria, 2024-07-07:
The webserver solution uses far less CPU time then the PULL method 5-8% against 20-40%
I still got the problem of NO devices and scenes in test branch. The rest of the screens work OK.
Tnx for the uptime.

Smanar, 2024-07-07:
>The webserver solution uses far less CPU time then the PULL method 5-8% against 20-40%

Yeah, realy not logic, but better for the device.

SargonofAssyria, 2024-07-07:
I had a tremendous lag in the device with test branch and PUSH method.
Might have something to do with those screens showing no devices and scenes.
Back to main branch from yesterday with PUSH and everything is fast and both screens are showing and are fast.
It is still a riddle to me what the cause might be.

SargonofAssyria, 2024-07-07:
So I started all over again on the TEST branch. Now it works as expected. With out the WS at the moment.
Will keep testing....

Smanar, 2024-07-07:
>With 1 exception, I added

```
;flash 16M
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.partitions = default_16MB.csv
```

You mean it work better or worse with it ? No more missing device on Home page ?
Sometime I delete the "pio" folder, to restart with something clean.

SargonofAssyria, 2024-07-08:
Currently the device works OK and is responsive. But after an OTA for some time the device is less responsive.
It looks like garbage collection kicks in, because after some time the device responds normal again.
No more missing devices on Home page.
I use default_16MB.csv because of more memory for OTA, but don't know if that is really true.
It does not seem to hurt.
What made you decide to use that layout with min_spiffs.csv apart from the info of the link, because that is about 4Mb devices.

Smanar, 2024-07-08:
>But after an OTA for some time the device is less responsive.

Yeah it can be normal, I have this problem using the USB upload too, after a reboot all is working fine.

>I use default_16MB.csv because of more memory for OTA, but don't know if that is really true.
It does not seem to hurt.
What made you decide to use that layout with min_spiffs.csv apart from the info of the link, because that is about 4Mb devices.

Depend of device, I have a esp32-2432S028R
**min_spiffs.csv**
https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/min_spiffs.csv
So it make 
code > 1 966 080
OTA > 1 966 080
SPIFF > 131 072

I m using this one because I don't use SPIFF (only few bytes) So I take the maximum memory for the OTA and the code (around 45% each).

**default_16MB.csv**
code > 6 553 600
OTA > 6 553 600
SPIFF > 3 538 944

So they don't look so different, apart you keep more memory for the SPIFF, but as you have more memory than me .....

I haven't reaction on domoticz yet, but I think I will use the PUSH mode. Less CPU use, not logic, but for me better for the device, if there is no performance lost. Perhaps it more unsecure, but I don't think someone can acess your network by hacking this kind of device.

SargonofAssyria, 2024-07-08:
I did several upload today with different configs all with the PUSH method, and you have to reboot sometimes.
I changed the height in the info_panel from 20 to 40 and font to font1. It works better on my device, because otherwise I cannot select a line (to small) At the final version those settings could perhaps be device depended.
Looks good now, will have to test more now with different device, Only tested with lights, temp, group, scene devices.
How do I change my Home page to the group page and visa versa? Because I think about working with scenes primary.

Smanar, 2024-07-08:
>I changed the height in the info_panel from 20 to 40

You mean for the button on top ? Or the table, Had same issue on the table with the new font, idk why but the padding is smaller.

`lv_obj_set_style_pad_all(table, 3, LV_PART_ITEMS);`

Have set the padding from 1 to 3.

>At the final version those settings could perhaps be device depended

If you have a capture (to see the result on your device) and the code you want I can make it permanent.

>How do I change my Home page to the group page and visa versa? 

In navigation.h try with

```
enum {
    TOOL_PANEL,
#ifndef NO_GROUP_PAGE
    GROUP_PANEL,
#endif
    HOMEPAGE_PANEL,
#if BONUSPAGE > 0
    BONUSPAGE_PANEL1,
#endif
#if BONUSPAGE > 1
    BONUSPAGE_PANEL2,
#endif
#if BONUSPAGE > 2
    BONUSPAGE_PANEL3,
#endif
#ifndef NO_INFO_PAGE
    INFO_PANEL,
#endif
    MAX_PANEL_SCROLL,
    DEVICE_PANEL,
    SETTING_PANEL
};
```

SargonofAssyria, 2024-07-08:
Yeah, The top row was to near to the edge to be conveniently pressed.
The table rows themselves might be a little more height too.
Oke tnx for the navigation info.
I will make some more adjustments then I will upload some screens.

SargonofAssyria, 2024-07-08:
Found what I meant -> in setup() ->     navigation_screen(GROUP_PANEL);

Smanar, 2024-07-08:
Ha yes, you mean at first page, not the order. So right you are at good place.
For the moment have removed the ESP2SOTA depedency, to be able to use the webserver for other thing later, it's perhaps a little more lighter.

SargonofAssyria, 2024-07-11:
I am happy with my ESP2SOTA, uploaded several tens of times. Always works, except you have to extra reboot the device every time after an update, otherwise next OTA update might not work or is not fluent.

Smanar, 2024-07-12:
>Always works, except you have to extra reboot the device every time after an update, otherwise next OTA update might not work or is not fluent

Yeah, and idk what to do, a timer before the reset ? Need to found similar issue.

SargonofAssyria, 2024-07-13:
No idea. I can live with the current situation, but people have to know that that can happen and what the solution is.

closed 2024-07-13 by SargonofAssyria