Smanar/CYD-Domoticz-Remote #13: Info panel is not showing all devices.

SargonofAssyria, 2024-07-06:
The info panel only showed the 'temp' devices.
After I changed the 'bar' into 'weather', I also got devices under weather.
The light and utility don't show any devices.
The output of the 4 json are:
```
light 162346 bytes
temp 24766 bytes
weather 13397 bytes
utility 128093 bytes
```
So I suspect that the size is the problem.
Is there a solution for this problem? Probably in the  HTTPGETRequestWithReturn function?

SargonofAssyria, 2024-07-06:
Found it, by setting it to false.
``
    if (!HTTPGETRequestWithReturn(buff, &doc, false)) return;
```

Smanar, 2024-07-06:
Ha ?
I think you have found something here.
The last flag is a filter, used to decrease size, so with a filter it work better on large size. If it work better with the filter set to false, it's probably because the filter remove too much stuff.

The filter is 
```
    // The filter: it contains "true" for each value we want to keep
    filter["result"][0]["Data"] = true;
    filter["result"][0]["idx"] = true;
    filter["result"][0]["Name"] = true;
```

I don't see something bad here, but I think, you are right with "streams", can you try with replacing

```
        if (NeedFilter)
        {
            err = deserializeJson(*doc, client.getString(), DeserializationOption::Filter(filter));
        }
```

by

```
        if (NeedFilter)
        {
            err = deserializeJson(*doc, client.getStream(), DeserializationOption::Filter(filter));
        }
```