Smanar/CYD-Domoticz-Remote #14: € sign is not good displayed on the screen.

SargonofAssyria, 2024-07-06:
The Euro sign (€) is displayed on the info screen as a box.
Also other sign like superscript 2 -> ² and mu -> μ
Any idea for a solution?

Smanar, 2024-07-06:
Yep it can depend of used font.
What is your device ? To see which one font it use.

By defaut on autogenerated one I m using as  special char `€°µ¥£²³` So you need to have them

SargonofAssyria, 2024-07-06:
I am using esp32-8048S070C

Smanar, 2024-07-06:
Exactly

```
    LV_FONT_DECLARE(Montserrat_Bold_18)
    const lv_font_t font1 = Montserrat_Bold_18;
    #ifdef CUSTOM_FONT
        LV_FONT_DECLARE(Montserrat_16)
        onst lv_font_t font3 = Montserrat_16;
        LV_FONT_DECLARE(Montserrat_14)
        const lv_font_t font2 = Montserrat_14;
    #else
        const lv_font_t font3 = lv_font_montserrat_16;
        const lv_font_t font2 = lv_font_montserrat_14;
    #endif
```

You still have CUSTOM_FONT to false ?
So  those one are native

```
       const lv_font_t font3 = lv_font_montserrat_16;
        const lv_font_t font2 = lv_font_montserrat_14;
```
And this one is custom 
`    const lv_font_t font1 = Montserrat_Bold_18;`

But this one have thoses special character
```
/*******************************************************************************
 * Size: 18 px
 * Bpp: 4
 * Opts: --bpp 4 --size 18 --no-compress --font Montserrat-Bold.ttf --symbols €°µ¥£²³ --range 32-126 --format lvgl -o Montserrat_Bold_18.c
 ******************************************************************************/
```

Do you have a capture to see wich one text miss it ?

SargonofAssyria, 2024-07-06:
[image]
Every where there is Prijs or Kosten there should be an € sign

SargonofAssyria, 2024-07-06:
I am using -DDEVICE_SIZE=3 # Device display size
So I should be on 
font1 = Montserrat_Bold_18;
font3 = lv_font_montserrat_16;
font2 = lv_font_montserrat_14;

SargonofAssyria, 2024-07-06:
So I tried font3 with Montserrat_Bold_18
That worked for € and other signs. But I think 16 and normal (not bold) would be better.
Side effect is that the firmware.bin is 10k **smaller**

Smanar, 2024-07-06:
Yeah, same for me it's smaller, have probably removed other special char.
So have pull a new version with generated font for Montserrat_12 and Montserrat_16.

I m not sure, but for me they are smaller too (not only space but size too). Was forced to increase padding on info panel.

Smanar, 2024-07-07:
Still have  the issue ?

SargonofAssyria, 2024-07-07:
No, with those 'new' fonts they where fine on the test branch, but for some reason the home panel did not display devices on the test branch, so I returned to master branch.

Smanar, 2024-07-07:
Can you check with "serial monitor" to see if you have some ouput ?
To see what is blocking, on my side with bad font, it freeze.

If you set 
`const lv_font_t font3 = Montserrat_Bold_18; `

Still the same issue ? (even you will have graphical issue with this edition)

SargonofAssyria, 2024-07-07:
I copied the Montserrat_16 font to my environment of yesterday and changed font3 to that font.
No problems and € and others are display fine. So it has to be something else... will keep searching...

SargonofAssyria, 2024-07-11:
This works fine for me now with test branch. Quick response and no crashes.
font1 = Montserrat_Bold_18
font2 = lv_font_montserrat_14
font3 = Montserrat_16
I am now testing with font2 = Montserrat_16 In my case the text just fits.
So This issue is solved.

closed 2024-07-11 by SargonofAssyria