Smanar/CYD-Domoticz-Remote #15: Light/Switch, Switch, Dimmer devices are not correctly displayed

SargonofAssyria, 2024-07-09:
When A lightbulb (zigbee in my case) has the level on 60% but is switchoff, the widget in the display is switched on.
I think the Status of the device should be reflected on the display and not the level.

Smanar, 2024-07-13:
It's like that the code need to work ....
```
    // Display a "on" icon
    if (d->type < TYPE_SWITCH && strcmp(d->data, "On") == 0)
    {
        lv_obj_t * label = lv_label_create(Button_icon);
        lv_obj_set_style_text_color(label, color, 0);
        lv_obj_align_to(label, img,  LV_ALIGN_OUT_RIGHT_BOTTOM, 0, 0);
        //lv_obj_set_style_border_width(label, 5, 0); // To make it visible
        lv_label_set_text(label, " On");
    }
```

Can you check the "data" field for this device ?
Using `http://192.168.1.1:8080/json.htm?type=command&param=getdevices&rid=XXX`

SargonofAssyria, 2024-07-13:
Tnx, it looks like this is working correct now. Probably solved with the WS change.

closed 2024-07-13 by SargonofAssyria