Smanar/CYD-Domoticz-Remote #16: Height of items in the Info panel

SargonofAssyria, 2024-07-11:
I already changed the height of the header from 20 to 40 because the header is at the top and not so good clickable if you have a case in which the display is placed.
The same applies to the rows.
It would be nice if that row height could be configurable.

SargonofAssyria, 2024-07-11:
I did a try with font3 = Montserrat_20 and
    lv_table_set_col_width(table, 1, 4*w/10);
    lv_table_set_col_width(table, 0, 6*w/10);

And that fine for me.

Smanar, 2024-07-11:
Have starting to make something for highter resolution n info_panel.cpp (have just updated the "test" branch).

```
#if DEVICE_SIZE == 1
    #define HEADERHEIGHT 20
    #define TABLEPAD 3
#else
    #define HEADERHEIGHT 40
    #define TABLEPAD 4
#endif
```

HEADERHEIGHT  you know what is it.
For the element table height, it's the lib that calculate it itself, according to padding and font size, so you can increase TABLEPAD 

For the table width you are using 40% and 60%, and by defaut it's 33% 66%, it make a so big difference ?

SargonofAssyria, 2024-07-11:
Tnx, the other col_width and font was just for experimenting 66% and 33% are fine, went back to font3 = Montserrat_16, that's fine too.
I updated with your defaults now, but used TABLEPAD 10 for my display just to get enough room for decent touch.

closed 2024-07-11 by SargonofAssyria

Smanar, 2024-07-12:
>I updated with your defaults now, but used TABLEPAD 10 for my display just to get enough room for decent touch.

Will be Updated on next version