Smanar/CYD-Domoticz-Remote #17: With Websockets device status is not ynchronized.

SargonofAssyria, 2024-07-12:
Using the latest beta build 16150 and compile test branch with -DLIGHTWS
The Status on the Domoticz light (On/Off) devices does not change on the display.
Compiled without #-DLIGHTWS the status is displayed correctly.

Smanar, 2024-07-13:
I can't reproduce it too.
I m trying with a bulb, and the "on" appear" close the icon if I turn it on.
Or you mean in scene/group panel ? here it's possible having issue, not tested yet.

SargonofAssyria, 2024-07-13:
It happens on the Home panel. I will try again and come back on this.

Smanar, 2024-07-13:
ATM I m editing the code to support other panel.

SargonofAssyria, 2024-07-13:
I switched back and forth on this setting with Websockets and Fastclic (Domoticz build 16150)
```
build_flags =
	-fexceptions
	-DPIO_FRAMEWORK_ARDUINO_ENABLE_EXCEPTIONS
	-DLV_CONF_PATH="../../../../src/conf/lv_conf.h" # file path
	-DUSER_SETUP_LOADED=1 # To enable config throught platformio for TFT_eSPI
	-DFORCE_CONFIG # Enable or disable the forced network configuration.
	-DFORCECALIBRATE=false # Reset the calibration at every restart, used to reset it after a config change.
	#-DOLD_DOMOTICZ # If you are using older Domoticz version < 2023.2
	-DBONUSPAGE=0 # Special page to add, similar to Homepage but not updated, from 1 to 3
	-DFASTCLIC # Enable 1 clic action for some widget
	#-DNO_INFO_PAGE # Disable the Info panel
	#-DNO_GROUP_PAGE # Disable the Group/scene panel
	#-DCUSTOM_FONT
	#-DAUTO_BRIGHTNESS
	
	
	-DLIGHTWS # Only possible for version > 16088
	-DPUSHOTA # To enable Push OTA (Don't enable both OTA)
	#-DPULLOTA # To enable PUSH OTA (Don't enable both OTA)
```
The results is that both settings work fine, but with the -DLIGHTWS the screen on the device stays on the original status of the device at startup

Smanar, 2024-07-13:
Have updated the code.
- Add a timer on OTA (1s) for test.
- Now scene/group panel can be updated too.

Have changed some stuff to WS but on my side.
- I turn a light "on".
- I have the "on" close the device on the HomePage
- I reset the device
- I have again the home page, with the "on" close the device.

The WS subscription stuff is new, perhaps I need to restart domoticz on my side (It have take around 20/30 subscription today ^^)

SargonofAssyria, 2024-07-13:
Yes, WebSocket is now working as it should.

Smanar, 2024-07-13:
And it's better for OTA on your side ?
Not better on mine.

SargonofAssyria, 2024-07-13:
That stop and delay on the OTA update did not work.
I will stick to the manual reset after every OTA. That works.
At the end updates will not be that frequent.

Smanar, 2024-07-13:
>I will stick to the manual reset after every OTA. That works.

You have an option to reset the device in the tool panel too.

SargonofAssyria, 2024-07-13:
Yes, that is the one I use every time after an OTA update. SO I am fine.

closed 2024-07-13 by SargonofAssyria