Smanar/CYD-Domoticz-Remote #18: Info screen cannot be swiped the first time when devices are displayed.

SargonofAssyria, 2024-07-12:
Working with Domoticz build 16150 and latest test branch. And compiled with -DFASTCLIC
When you are on the info screen and first selected lights and want to swipe back to an other screen, then first the touched device with the swipe is selected, then you go back, get an empty screen and then you can swipe back to an other screen.

Smanar, 2024-07-12:
Hu ? Can you describe the procedure step by step ? sorry I m not sure to understand

1 - go in info screen
2 - swipe back to home panel ?

For info I will update my domoticz version this WE for the other issue.

SargonofAssyria, 2024-07-13:
1. You swipe to the info screen.
2. You select lights and get an overview of all the lights.
3. At that moment it is (almost) impossible to swipe back to the scene panel without first a light being selected.
4. You get into the device 'sub-screen', than click Back to an empty screen,
5. Swipe back to the scene panel.

Some it goes good, but most of the time it goes wrong. The screen seems too sensitive, not much difference between the swipe and a click.
I don't know it it can be solved, it is not a very big issue, but something I noticed and can be annoying.

Smanar, 2024-07-13:
I can't reproduce it on my side.
- You swipe to the info screen.
- You select lights and get an overview of all the lights.
- I select a "light" and keep pressed
- The "light" is selected, (the background is changing)
- I make a swipe
- I return to "group/scene" tab

SargonofAssyria, 2024-07-13:
Sometimes I manage that too, but most of the time it's processed as a click.
Probably sensitivity on my Hardware, not a big issue, because I won't use that screen much.

closed 2024-07-13 by SargonofAssyria