Smanar/CYD-Domoticz-Remote #19: It would be nice to transfer all kind of settings to something like 'Setting.h'

SargonofAssyria, 2024-07-13:
I already use FORCE_CONFIG and put fixed IP in wifi_setup.cpp like
```
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    wifi_Connecting_screen();

    // Set your Static IP address
    IPAddress local_IP(192, 168, y, xxx);
    // Set your Gateway IP address
    IPAddress gateway(192, 168, y, xxx);
    IPAddress subnet(255, 255, 255, 0);
    IPAddress primaryDNS(192, 168, y, zzz);
    IPAddress secondaryDNS(8, 8, 8, 8);
    // Configures static IP address
    if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
        Serial.println("Failed to configure Static IP");
    } else {
        Serial.println("Static IP configured!");
    }

    WiFi.begin(global_config.wifiSSID, global_config.wifiPassword);
```
So it would be nice if all those and other settings could be transfered to just 1 file.

Smanar, 2024-07-13:
You mean in an external file, not updated when updating code ?

SargonofAssyria, 2024-07-13:
Yes, I do it with all my code.
The Setting.h is initially transfered from Github, but afterwards not synchronized by a pull.

Smanar, 2024-07-13:
Something like

```
    // Personnal Settings to don't have to set them at every reset.
    // They are saved after have been set, so the flag FORCE_CONFIG can be disabled after
    #if FORCE_CONFIG

        #if __has_include("personnal_settings.h")
            #include "personnal_settings.h"
        #else

            strcpy(global_config.wifiPassword, "xxxxxxxxxxxxxxxxxx");
            strcpy(global_config.wifiSSID, "xxxxxxxxxxxxxxxxx");
            strcpy(global_config.ServerHost, "192.168.1.1");
            global_config.ServerPort = 8080;
            global_config.wifiConfigured = true;
            global_config.ipConfigured = true;

            const static unsigned short t[] = {37, 75, 16, 36, 28, 35, 57, 89, 45};
            for (int i=0; i<TOTAL_ICONX*TOTAL_ICONY; i++)
                global_config.ListDevices[i] = t[i];

        #endif

        WriteGlobalConfig();
    #endif
```

SargonofAssyria, 2024-07-13:
Yes, that would be perfect, and also for the other setting like fixed IP.

Smanar, 2024-07-13:
I think for your "special stuff better for you to use
```

    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    wifi_Connecting_screen();

        #if __has_include("personnal_settings.h")
            #include "personnal_settings.h"
            special_settings();
        #endif
    WiFi.begin(global_config.wifiSSID, global_config.wifiPassword);
```

And in "personnal_settings.h"
```
void special_settings(void)
{
    // Set your Static IP address
    IPAddress local_IP(192, 168, y, xxx);
    // Set your Gateway IP address
    IPAddress gateway(192, 168, y, xxx);
    IPAddress subnet(255, 255, 255, 0);
    IPAddress primaryDNS(192, 168, y, zzz);
    IPAddress secondaryDNS(8, 8, 8, 8);
    // Configures static IP address
    if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
        Serial.println("Failed to configure Static IP");
    } else {
        Serial.println("Static IP configured!");
    }
}
```

Tell me if it work, I will add it in code.

SargonofAssyria, 2024-07-13:
No I got this error.
```
src/ui/personal_settings.h:2:1: error: a function-definition is not allowed here before '{' token
Compiling .pio\build\esp32-8048S070C\libd66\lvgl\draw\sdl\lv_draw_sdl_rect.c.o
```
Can't figure out where the error is.

Smanar, 2024-07-13:
A wrong number of "{" or "}".
Can you share your personal_settings.h file ?

And I m seeing need to choose a path too, probably more logic to put in same directory than the main.cpp ?

SargonofAssyria, 2024-07-13:
May be you could use something in the "personal_settings.h" (just 1 n in personal) like

#define USE_FIXEDIP = 1
#ifdef USE_FIXEDIP
#define ST_IP      192,168,y,xxx
#define ST_GATEWAY 192,168,y,xxx 
#define ST_NETMASK 255,255,255,0
// One or two optional DNS servers can be supplied.
#define ST_DNS1 192,168,y,xxx
#define ST_DNS2 8,8,8,8

#endif

And that define also in the source with the right code.

SargonofAssyria, 2024-07-13:
```
void special_settings(void)
{
    // Set your Static IP address
    IPAddress local_IP(192, 168, 2, 95);
    // Set your Gateway IP address
    IPAddress gateway(192, 168, 2, 1);
    IPAddress subnet(255, 255, 255, 0);
    IPAddress primaryDNS(192, 168, 2, 248);
    IPAddress secondaryDNS(8, 8, 8, 8);
    // Configures static IP address
    if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
        Serial.println("Failed to configure Static IP");
    } else {
        Serial.println("Static IP configured!");
    }
}
```
The {} fit as far as I can see.

Smanar, 2024-07-13:
I have used your previous comment, new version online
The file need to be in folder "src", I think you will understand how to use it
```

    #if FORCE_CONFIG

        #if __has_include("personnal_settings.h")
            #include "personnal_settings.h"

            strcpy(global_config.wifiPassword, WIFIPASSWORD);
            strcpy(global_config.wifiSSID, WIFISSID);
            strcpy(global_config.ServerHost, SERVERHOST);
            global_config.ServerPort = SERVERPORT;

        #else
            strcpy(global_config.wifiPassword, "cexZJXZC41xrP3fkmU9");
            strcpy(global_config.wifiSSID, "Bbox-Bruce");
            strcpy(global_config.ServerHost, "192.168.1.1");
            global_config.ServerPort = 8080;
        #endif

        global_config.wifiConfigured = true;
        global_config.ipConfigured = true;

        const static unsigned short t[] = {37, 75, 16, 36, 28, 35, 57, 89, 45};
        for (int i=0; i<TOTAL_ICONX*TOTAL_ICONY; i++)
            global_config.ListDevices[i] = t[i];

        WriteGlobalConfig();
    #endif
```
```

    // Special config ?
    #if __has_include("../personnal_settings.h")
        #include "../personnal_settings.h"

        #ifdef USE_FIXEDIP
            IPAddress staticIP(ST_IP);
            IPAddress gateway(ST_GATEWAY);
            IPAddress subnet(ST_NETMASK);
            WiFi.hostname(WIFI_HOSTNAME);
            #if !defined(ST_DNS1)
                WiFi.config(staticIP, gateway, subnet);
            #else
                IPAddress dns1(ST_DNS1);
                #if !defined(ST_DNS2)
                    WiFi.config(staticIP, gateway, subnet, dns1);
                #else
                    IPAddress dns2(ST_DNS2);
                    WiFi.config(staticIP, gateway, subnet, dns1, dns2);
                #endif
            #endif
        #endif
    #endif
```

SargonofAssyria, 2024-07-13:
Change/Edit your post above. password

SargonofAssyria, 2024-07-13:
Yes my personal_settings.h is now  (mind the single 'n' e.g. English)
```
#define WIFI_HOSTNAME "ZZZ"
#define SERVERHOST  "192.168.y.xxx"
#define SERVERPORT 8080

#define WIFISSID  "<ssid>"
#define WIFIPASSWORD  "<password>"

#define USE_FIXEDIP = 1

#ifdef USE_FIXEDIP
    #define ST_IP 192,168,y,xxx
    #define ST_GATEWAY 192,168,y,xxx
    #define ST_NETMASK 255,255,255,0
    // One or two optional DNS servers can be supplied.
    #define ST_DNS1 192,168,y,xxx
    #define ST_DNS2 8,8,8,8
#endif
```

SargonofAssyria, 2024-07-13:
You could also remove the #else part in main.cpp and also transfer the
```
const static unsigned short t[] = {37, 75, 16, 36, 28, 35, 57, 89, 45};
```
to the personal_settings.h

SargonofAssyria, 2024-07-13:
Used it, compiled it, it works perfect....

closed 2024-07-13 by SargonofAssyria