Smanar/CYD-Domoticz-Remote #20: Device detail information.

SargonofAssyria, 2024-07-13:
The detail information of a device for instance Temperature could be possibly be better arranged.
The header where the devicename (and icon) could be much less in height.
That way the graph could increase in height (more clear info).
It would be nice if the x-as is named, because it looks to me that the x-as is 1 day, so every line is 6 hours.
Would even be better if those x-as could be with 24-hours as values.

Smanar, 2024-07-13:
Can you share a picture ?
I have a smaller resolution 240*480
Value used in bigger resolution are proportionnal or give by others users.

And the X axis is always 24h.
I don't name it to have more place as the value is fixed.

SargonofAssyria, 2024-07-13:
Sure.
[image]
The real display is way better then this photo.
But you can see the white space on the Devicename and the setting of 
```
  lv_obj_set_size(cont, lv_pct(90), lv_pct(90));
```
Works on this panel too, for me there is no need to set it to 80.

Smanar, 2024-07-13:
Better with ?

```
void device_panel_init(lv_obj_t* panel)
{

    lv_color_t color = Getcolor(SelectedDevice->type);
    const lv_img_dsc_t *icon = Geticon(SelectedDevice->type);
    lv_obj_t * label;

    /*Create a container with grid*/
    lv_obj_t * cont = lv_obj_create(panel);

#if DEVICE_SIZE == 1
    static lv_coord_t col_dsc[] = {LV_GRID_FR(34), LV_GRID_FR(33), LV_GRID_FR(33), LV_GRID_TEMPLATE_LAST};
    static lv_coord_t row_dsc[] = {LV_GRID_FR(30), LV_GRID_FR(54), LV_GRID_FR(16), LV_GRID_TEMPLATE_LAST};
    lv_obj_set_grid_dsc_array(cont, col_dsc, row_dsc);
    lv_obj_set_size(cont, lv_pct(90), lv_pct(90));
#else
    static lv_coord_t col_dsc[] = {LV_GRID_FR(34), LV_GRID_FR(33), LV_GRID_FR(33), LV_GRID_TEMPLATE_LAST};
    static lv_coord_t row_dsc[] = {LV_GRID_FR(20), LV_GRID_FR(64), LV_GRID_FR(16), LV_GRID_TEMPLATE_LAST};
    lv_obj_set_grid_dsc_array(cont, col_dsc, row_dsc);
    lv_obj_set_size(cont, lv_pct(80), lv_pct(80));
#endif
    //lv_obj_set_size(cont, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
    lv_obj_center(cont);
```

SargonofAssyria, 2024-07-13:
Yes, that's it.
[image]

closed 2024-07-13 by SargonofAssyria