Smanar/CYD-Domoticz-Remote #21: Icon size.

SargonofAssyria, 2024-07-13:
You save my pictures. I would like to have the icons a little bit bigger like 60x60
Could you have one of the originals transform to that size so I can test?
[Wish] Even better would be that some of the icons are in 2 colors, like switches, lights, warnings and plugs. 1 color for each state.

Smanar, 2024-07-14:
We have already tried with bigger icon, you have a file to test here for temperature https://github.com/Smanar/CYD-Domoticz-Remote/blob/eba139a4198a53a7b5ec0c91cc77506080a06f35/src/image/temperature60x60.c

But we have give up, because on bigger resolution, better having more data than bigger icon. He was using 4*3 icons if I remember.

If you want I can explain how to make the file.

>Even better would be that some of the icons are in 2 colors, like switches, lights, warnings and plugs. 1 color for each state.

I m using monochrome icon, all black, and I make operation on them, color, alpha blending. So they are using only 1 color.
I can change color or transparency without problem, but not possible use 2 colors on the same icon on memory, because the recolorisation will be random, and will take more memory.

We have tried some graphical thing, like adding circle, boxes, but results was not better.

SargonofAssyria, 2024-07-14:
Tnx for the 60x60 icon. I think it looks better, but that is just me. B.t.w. This icon has no transparent background (yet).
I would like to pursue this further.
Please explain how to make this and other icon files.

Smanar, 2024-07-14:
I m using image like this one, monochrome + transparent.
[image]

Then I go to https://lvgl.io/tools/imageconverter
- Version : Lvlg v8
- format : CF_ALPHA_1_BIT
- Output : C array

SargonofAssyria, 2024-07-17:
Tnx for the info, experimented already a little.
No good result yet. Will keep on testing....

closed 2024-07-17 by SargonofAssyria