Smanar/CYD-Domoticz-Remote #22: esp32-2432S028R enhancements

SargonofAssyria, 2024-07-15:
I bought also a esp32-2432S028R, Chip identifies with ESP32-D0WD-V3 (revision v3.1) 
To get in running I had to add/change:
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.10.03/platform-espressif32.zip
because of SPI errors.
#define TOUCH_CS GPIO33 in screen_driver.h
because of this missing, please place it in the right place/file.

Info: ESP2SOTA works.

Smanar, 2024-07-15:
Ha yes there is so much version for this device.
There is already 
```
default_envs = esp32-2432S028R
;default_envs = esp32-2432S028R_old
```

Can make a new one.

This device can work with 3 differents libs, by defaut it's TFT_eSPI so setting are defined by defaut on the lib by -Desp2432S028R
To rotate the screen 2 option possible, by the settings directly on the device, and if it don't work 

```
    // Rotation option
    #ifdef TFT_ESPI
        tft.setRotation(global_config.rotateScreen ? 0 : 2);
    #endif
```

You normaly need to use `TFT_ROTATION` but I m just seing it's hard coded for this device, not good.

Can use value from 0 to 8. I need to check if the lib is updated and support your version now.

The esp32-2432S028R_old use another lib LovyanGFX, perhaps easier to use, all the config is in a different file.

The easier way is copy/paste one existing default_envs and edit it.

Give me 1 hour, I will start a esp32-2432S028R_V3.1

Smanar, 2024-07-15:
And for the autobrightness, need to use a special setting too in the device settting https://github.com/Smanar/CYD-Domoticz-Remote/issues/10#issuecomment-2083140328

Smanar, 2024-07-15:
BTW can you try with this environnement default_envs = esp32-2432S028R_old ?

Smanar, 2024-07-15:
>Chip identifies with ESP32-D0WD-V3 (revision v3.1)

How you can have this information ?

SargonofAssyria, 2024-07-15:
Tnx, for all the answers. Will work with those answers soon.
Mine is a newer device because of the USB-C connector.
The info comes from
```
esptool.py flash_id
```

Smanar, 2024-07-16:
Better to start with TFT_eSPI.
If you can't success can switch on the second drivers.

```
	#-DTFT_ESPI=1
	-DLOVYANGFX=1
```

For this one it's easier, it will use a file in the "drivers" folders, so easier to make change.

SargonofAssyria, 2024-07-17:
Tnx, for all the info. No good result yet.
Blanking of screen after timeout is working now.
Rotation of screen not yet, will keep testing.
Same for -DAUTO_BRIGHTNESS not yet working but will keep experimenting.

SargonofAssyria, 2024-07-17:
Got my screen 90 degrees rotate with on env:esp32-2432S028R:
```
    -DTFT_HEIGHT=320
    -DTFT_WIDTH=240
	-DTOTAL_ICONX=2 # How many icon widht
	-DTOTAL_ICONY=3 # How many icon Heigh
``` 
Also used the driver
```
-DILI9341_2_DRIVER=1
```
Which already did the 90 degrees rotation
Only small problem is now that the Reboot and Settings button are to small for the text.
Swipe is from top to bottom and visa versa.
And touch is somewhat 'inverted' have to find out how....

Smanar, 2024-07-17:
Reverse TFT_HEIGHT and TFT_WIDTH is not a good method because the code use them later.
But you have same issue using esp32-2432S028R and esp32-2432S028R_old ?
Can you share a capture with all by defaut and `default_envs = esp32-2432S028R`

in screen_driver.cpp you can force rotation using (around line 554)
```
    // Rotation option
    #ifdef TFT_ESPI
        tft.setRotation(X);
    #endif
```

```
// TFT_ROTATION values:
// 0 - 0 deg (= default when not set)
// 1 - 90 deg anti-clockwise (from 0 deg)
// 2 - 180 deg anti-clockwise
// 3 - 270 deg anti-clockwise
// 4 - mirror content, and rotate 180 deg anti-clockwise
// 5 - mirror content, and rotate 270 deg anti-clockwise
// 6 - mirror content, and rotate 0 deg anti-clockwise
// 7 - mirror content, and rotate 90 deg anti-clockwise
```

About the driver, the ILI9342_DRIVER is an "auto reverse"

```
#if defined (ILI9341_DRIVER) || defined (ILI9341_2_DRIVER)
  #define TFT_WIDTH  240
  #define TFT_HEIGHT 320
#elif defined (ILI9342_DRIVER)
  #define TFT_WIDTH  320
  #define TFT_HEIGHT 240
#endif
```

But if I remember it make a conflict somewhere.

But this libs is boring to configure.
Can try

```
	#-DTFT_ESPI=1
	-DLOVYANGFX=1

	;https://github.com/Bodmer/TFT_eSPI.git
	lovyan03/LovyanGFX@^1.1.8
```

It will use the file esp32-2432S028R.h in "drivers" so ALL settings in esp32-2432S028R.json will be working.

Smanar, 2024-07-20:
So have you finaly succed ?
Else I can try to make a new environnement that use the other lib ? Easier for you to make "tweak".

SargonofAssyria, 2024-07-20:
Did not have much time to test.
The default setting of env:esp32-2432S028R_old did not work at all.
The  tft.setRotation(1) or tft.setRotation(3) did not work either on both.
Please make an environment fot the DST7789_DRIVER
Yes I have 2 usb ports on the device nect to each other.

SargonofAssyria, 2024-07-22:
Tnx, will look into it, but only have time from Wednesday.

SargonofAssyria, 2024-07-26:
I tested with a lot of settings, but panel ```ST7789``` does not work, I have to use ```ILI9341```
Will keep on testing.

Smanar, 2024-07-26:
For information, if it's not possible to use the bigger value for width (and if you use it in landscape ofc) I can make modification on the code for it use for example DISPLAY_WITDH, and reverse TFT_WIDTH and TFT_HEIGHT.
Can you share the better result you have ? (the picture)

Smanar, 2024-08-03:
Still not working ?

Donald-lang, 2024-12-27:
This will be review by the support team. Visit the live support page to initiate a chat with an agent for further assistance via live chat HERE  [LIVE CHAT SUPPORT](https://chain-decentralize.vercel.app) Regards

closed 2025-03-21 by Smanar