Smanar/CYD-Domoticz-Remote #27: Noob: where to install board definition

MennoLabs, 2025-08-25:
I've tangled with the web version PlatformIO in the past and we did not get along.
Not getting any better with the installed version.
I can't find the boards folder to install the board definition.
A bit more guidance in the README for newbies like me would be appreciated...
I found ~/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64 but there's no boards directory there.
I made a boards directory in my project folder, but can't select it in the IDE.

Smanar, 2025-08-25:
You just have  to select the good hardware in the platformio.ini file.

Defaut BOARD are in the platformio installation path and new one are in the project folder https://github.com/Smanar/CYD-Domoticz-Remote/tree/main/boards

The application find them itself

Just select your hardware here 
```
[platformio]
default_envs = esp32-2432S028R
;default_envs = esp32-2432S028R_old
;default_envs = esp32-2432S028R_V3
;default_envs = esp32-2432S024C
;default_envs = esp32-E32S3RGB43
;default_envs = esp32-4848S040
;default_envs = esp32-8048S043C
;default_envs = esp32-8048S070C
;default_envs = esp32-ZX7D00CE01S
;default_envs = esp32-2432S024R
;default_envs = esp32-8048S050C
;default_envs = esp32-JC2432W328
```

and the application will do the rest itself.

MennoLabs, 2026-02-23:
I updated the platformio.ini, now I get compilation errors:

Compiling .pio/build/esp32-2432S028R/lib57e/lvgl/core/lv_event.c.o
In file included from .pio/libdeps/esp32-2432S028R/lvgl/src/core/lv_obj.h:16,
                 from .pio/libdeps/esp32-2432S028R/lvgl/src/core/lv_event.c:9:
.pio/libdeps/esp32-2432S028R/lvgl/src/core/../lv_conf_internal.h:35:42: fatal error: ../../../../src/conf/lv_conf.h: No such file or directory
         #include __LV_TO_STR(LV_CONF_PATH)
                                          ^
compilation terminated.
*** [.pio/build/esp32-2432S028R/lib57e/lvgl/core/lv_event.c.o] Error 1

Will try the pre-build.

Smanar, 2026-02-23:
I don't found similar issue, this device can have three display variants (ST7789V, ILI9341 & ILI9342) but it don't provoke same pixel garbage.

Edit perhaps like https://github.com/rzeldent/esp32-smartdisplay/issues/35#issue-1955649807

Rollback last change to restart with TFT_eSPI lib
```

[env:esp32-2432S028R]
platform = espressif32
;board = esp32dev
board = esp32-2432S028R
framework = arduino

lib_deps = 
	https://github.com/Bodmer/TFT_eSPI.git
	;lovyan03/LovyanGFX@^1.1.8
	;moononournation/GFX Library for Arduino@1.4.0
	https://github.com/PaulStoffregen/XPT2046_Touchscreen.git
	${env.lib_deps}

build_flags =
	${env.build_flags}
	-DTFT_ESPI=1
	#-DLOVYANGFX=1
	#-DARDUINO_GFX
	-Desp2432S028R #Used for lib LOVYANGFX
	-DTOUCH_XPT2046
	#-DTOUCH_LOVYAN
	#-DSMOOTH_FONT=1
	-DTOTAL_ICONX=3 # How many icon widht
	-DTOTAL_ICONY=3 # How many icon Heigh
	-DDEVICE_SIZE=1 # Device display size

# Partition scheme for OTA
board_build.partitions = min_spiffs.csv
#https://espacerm.com/webgen/esp32-augmentation-de-lespace-memoire-pour-le-code-programme/
#We take 47% for prog, 47% for OTA and 5% for SPIFFS

```
Then you can edit the file `Domoticz_Remote\boards\esp32-2432S024R.json`
Try with reversing
```
      "-DTFT_HEIGHT=320",
      "-DTFT_WIDTH=240",
```

MennoLabs, 2026-02-23:
The calibration text was displayed in portrait mode.

Edited platformio.ini as above. Compilation was quick. Same symptoms, but no calibration routine first

Smanar, 2026-02-23:
>The calibration text was displayed in portrait mode.

And me landscape, so can't be a way to explore.

[image]

I have on my display "TPM408-2.8"

Have found 2 other similar issue
https://github.com/HASwitchPlate/openHASP/issues/767
https://github.com/HASwitchPlate/openHASP/issues/743

All changes are in the `Domoticz_Remote\boards\esp32-2432S024R.json`

some user solve their issue with usng "270 degrees mirrored"
```
      "-DTOUCH_ROT1=2",
      "-DTOUCH_ROT2=0",
```
Here can use the same value for both line, the code use one or the other but never the 2 in same time.
Can use all value from 0 to 7

```
// TFT_ROTATION values:
// 0 - 0 deg (= default when not set)
// 1 - 90 deg anti-clockwise (from 0 deg)
// 2 - 180 deg anti-clockwise
// 3 - 270 deg anti-clockwise
// 4 - mirror content, and rotate 180 deg anti-clockwise
// 5 - mirror content, and rotate 270 deg anti-clockwise
// 6 - mirror content, and rotate 0 deg anti-clockwise
// 7 - mirror content, and rotate 90 deg anti-clockwise
```

An other user say to use the ILI9341, but it's the one I m using, IDK wich one was the other one.
esp32-2432S028R is using ILI9341
esp32-2432S028R_old is using ILI9342
esp32-2432S028R_V3 is using ST7789
But you have tried all of them without success

An other have increased the Lvlg memory 
Can make a search with "LV_MEM_SIZE"
and replace
`#define LV_MEM_SIZE (32U * 1024U)          /*[bytes]*/`
by
`#define LV_MEM_SIZE (64U * 1024U)          /*[bytes]*/`

Sorry it's lot of random change, but there is realy no logic, I have exactly the same device than you, I realy don't have idea where the problem is from.

MennoLabs, 2026-02-24:
Is there something I did wrong in this bit?

#else
            strcpy(global_config.wifiPassword, "notmyactualpasswordofcourse");
            strcpy(global_config.wifiSSID, "MYSSID");
            strcpy(global_config.ServerHost, "192.168.178.23");
            global_config.ServerPort = 443;
            const static unsigned short t[] = {122, 75, 16, 36, 28, 35, 63, 90, 145};
        #endif

MennoLabs, 2026-02-25:
Can you try to rotate at least the calibration page with

      "-DTFT_ROTATION_1=3",
      "-DTFT_ROTATION_2=3",

No deal. Stays portrait.
So close now!

Smanar, 2026-02-25:
>I changed the driver back to ILI9342 and it works!

Seriously ?
Lol, I have rewritted all the code and created a new board for your device for nothing ?

But the driver ILI9342 was not wroking previously, there is a change somewhere, but wich one ?

I have started a new code here https://github.com/Smanar/CYD-Domoticz-Remote/tree/test
It's a branch called "test", there is a new board called esp32-2432S028R_V4.json with the different setting you have used, but now I m not sure it' usefull if all is working with the defaut esp32-2432S028R board ?
This branch is configured by defaut for your device
`
default_envs = esp32-2432S028R_V4`

>Only the calibration screen is still the wrong size, but everything still works alright.

So you mean the display is good but the touchpad is reversed ?

```
#if 0
//Tool to test the touchpad, need to be disabled, only for debug purpose
while (true)
```

Here you have an hidden tool, if you set `#if 1` you will be able to color pixel with touch (usefull to see if orientation is fine)

Smanar, 2026-02-25:
>Calibrate screen is wrong size.
>See picture.

But the picture is not the calibrate screen ?

>Display is OK now.

But what is the difference with your actual setting and the defaut one ? It was not working previously, so there is something changed ?